FOSSASystems/FOSSASAT-1 #2: PDFs of PCB designs are not source code

Snowda, 2019-08-16:
PDFs aren't even usable with manufacturers. That would require the gerber files.

But even then, the hardware is not currently open source without the original Altium/Kicad/Eagle/etc. files that can be modified or improved upon by outside individuals.

marzoukali, 2019-08-22:
I was concerned about the same point, I want to make a big contribution there and i went through the whole code but i was wonder if we can create an easy integration test so we can solve the hardware closed source problem.

juliantech5, 2019-09-07:
Hey all, excuse the delay. Been so busy getting the satellite built and tested that I have not had the chance to update. I will upload the gerber files of the last models and the latest source code soon.

vinicentus, 2019-12-04:
Is there a timeline for this?

juliantech5, 2019-12-04:
uploaded latest gerbers

closed 2019-12-04 by juliantech5

peterus, 2019-12-14:
wanted to check with JLCPCB if everything is here and what it would cost to order some PCBs with SMT components added to the PCBs. Looks like the Pick&Place File is missing. @juliantech5 can you please upload the file too? Thanks!