FOSSASystems/FOSSASAT-1 #4: Error compiling: not declared in this scope

K4KDR, 2019-11-04:
Thank you for the updates to the Arduino code for FossaSat-1!

Using commit 70ef487, I get the following error.  Please note that 'FOSSA-Comms' is up-to-date as of the current commit which is 1fc86e2 at the date of this post.

Thank you very much!

-Scott,  K4KDR

```
/home/k4kdr/Arduino/libraries/FOSSA-Comms/src/FOSSA-Comms.cpp: In function 'int16_t FCP_Get_OptData_Length(char*, uint8_t*, uint8_t, const uint8_t*, const char*)':
/home/k4kdr/Arduino/libraries/FOSSA-Comms/src/FOSSA-Comms.cpp:57:5: error: 'aes128_dec_multiple' was not declared in this scope
     aes128_dec_multiple(key, encSection, encSectionLen);
     ^~~~~~~~~~~~~~~~~~~
/home/k4kdr/Arduino/libraries/FOSSA-Comms/src/FOSSA-Comms.cpp: In function 'int16_t FCP_Get_OptData(char*, uint8_t*, uint8_t, uint8_t*, const uint8_t*, const char*)':
/home/k4kdr/Arduino/libraries/FOSSA-Comms/src/FOSSA-Comms.cpp:141:5: error: 'aes128_dec_multiple' was not declared in this scope
     aes128_dec_multiple(key, encSection, encSectionLen);
     ^~~~~~~~~~~~~~~~~~~
/home/k4kdr/Arduino/libraries/FOSSA-Comms/src/FOSSA-Comms.cpp: In function 'int16_t FCP_Encode(uint8_t*, char*, uint8_t, uint8_t, uint8_t*, const uint8_t*, const char*)':
/home/k4kdr/Arduino/libraries/FOSSA-Comms/src/FOSSA-Comms.cpp:245:5: error: 'aes128_enc_multiple' was not declared in this scope
     aes128_enc_multiple(key, encSection, encSectionLen + paddingLen);
     ^~~~~~~~~~~~~~~~~~~
Multiple libraries were found for "RadioLib.h"
 Used: /home/k4kdr/Arduino/libraries/RadioLib
Multiple libraries were found for "SPI.h"
 Used: /home/k4kdr/Downloads/arduino/arduino-1.8.10/hardware/arduino/avr/libraries/SPI
Multiple libraries were found for "SoftwareSerial.h"
 Used: /home/k4kdr/Downloads/arduino/arduino-1.8.10/hardware/arduino/avr/libraries/SoftwareSerial
Multiple libraries were found for "FOSSA-Comms.h"
 Used: /home/k4kdr/Arduino/libraries/FOSSA-Comms
Multiple libraries were found for "AESLib.h"
 Used: /home/k4kdr/Arduino/libraries/AESLib
Using library RadioLib at version 1.7.0 in folder: /home/k4kdr/Arduino/libraries/RadioLib 
Using library SPI at version 1.0 in folder: /home/k4kdr/Downloads/arduino/arduino-1.8.10/hardware/arduino/avr/libraries/SPI 
Using library SoftwareSerial at version 1.0 in folder: /home/k4kdr/Downloads/arduino/arduino-1.8.10/hardware/arduino/avr/libraries/SoftwareSerial 
Using library FOSSA-Comms at version 1.0.0 in folder: /home/k4kdr/Arduino/libraries/FOSSA-Comms 
Using library AESLib at version 1.0.5 in folder: /home/k4kdr/Arduino/libraries/AESLib 
exit status 1
Error compiling for board Arduino/Genuino Uno.
```

jgromes, 2019-11-04:
Judging by this:

```
Multiple libraries were found for "RadioLib.h"
 Used: /home/k4kdr/Arduino/libraries/RadioLib
Multiple libraries were found for "SPI.h"
 Used: /home/k4kdr/Downloads/arduino/arduino-1.8.10/hardware/arduino/avr/libraries/SPI
Multiple libraries were found for "SoftwareSerial.h"
 Used: /home/k4kdr/Downloads/arduino/arduino-1.8.10/hardware/arduino/avr/libraries/SoftwareSerial
Multiple libraries were found for "FOSSA-Comms.h"
 Used: /home/k4kdr/Arduino/libraries/FOSSA-Comms
Multiple libraries were found for "AESLib.h"
 Used: /home/k4kdr/Arduino/libraries/AESLib
```

there's something wrong with your libraries or Arduino IDE installation/configuration. Please make sure Arduino IDE and all libraries are installed correctly.

K4KDR, 2019-11-04:
Thank you very much for the guidance!

Arduino IDE was re-installed and FOSSASystems repositories & required libraries re-downloaded. Pulled commit a5fddca; code verifies & uploads without error.

Many thanks for your support!

closed 2019-11-05 by jgromes