FOSSASystems/FOSSASAT-1 #5: Add "AESLib" to the list of required libraries?

K4KDR, 2019-11-04:
Using commit a5fddca, the following error appears on VERIFY:

```
/home/k4kdr/Arduino/libraries/FOSSA-Comms/src/FOSSA-Comms.h:5:10: fatal error: AESLib.h: No such file or directory
 #include <AESLib.h>
```

... after installing AESLib, (manually from https://github.com/DavyLandman/AESLib), the code will VERIFY & UPLOAD without error.

You may wish to add library "AESLib" to the list of required libraries if that is correct.

Thank you!

-Scott,  K4KDR

jgromes, 2019-11-05:
Arduino IDE has no way to install dependencies automatically, you always have to install all required libraries manually.

Added links to AESLib to [FOSSA-Comms readme](https://github.com/FOSSASystems/FOSSA-Comms/blob/master/README.md) and [Comms Guide](https://github.com/FOSSASystems/FOSSASAT-1/blob/master/FOSSA%20Documents/FOSSASAT-1%20Comms%20Guide.pdf), hopefully it's clear now.

closed 2019-11-05 by jgromes

K4KDR, 2019-11-05:
Thank you!

Yes, it will certainly help others to have a clear list of dependencies.