FOSSASystems/FOSSASAT-1 #6: Retransmitted message can contain additional characters

K4KDR, 2019-11-28:
Question or possible issue ref. the command:

`r - send message to be retransmitted`

When a simple string is transmitted, the Simulator reply is initially as expected:

```
00:03:38.314 -> Got repeated message:
00:03:38.366 -> hello
```

... but when you send additional messages, the reply can include extra characters:

```
00:05:05.545 -> Got repeated message:
00:05:05.579 -> hello again⸮⸮⸮&
```

... and the really strange reply comes when a short message is sent AFTER a longer message:

```
00:07:20.852 -> Got repeated message:
00:07:20.852 -> 123456789_123456789_12345ds⸮⸮⸮

00:08:01.194 -> Got repeated message:
00:08:01.229 -> hi there9_12345
```

... perhaps is something in either the GroundStation code or the Simulator code not clearing out a buffer between re-transmitted messages?  Also, in certain circumstances, the re-transmission from the simulator dumps dozens or hundreds of characters - I assume an overflow condition of some kind?

Of course this is not a big deal and I only bring it to your attention in case the spacecraft code behaves like the simulator code in this regard.  NOT a priority, obviously; just wanted to share my observation as we anxiously await the launch.

Great to see so many people registered on the Ground Station site now!

-Scott,  K4KDR

jgromes, 2019-11-28:
Hi Scott, thanks for pointing this out.

In the packet dumps, you can see that the data actually sent over the air is correct. There was an issue  with how the received data is printed in the ground station code - it basically just printed everything in RAM until it hit null byte, which is why you got seemingly random bytes at the end of the printed string.

Should be fixed now.

closed 2019-11-28 by jgromes

K4KDR, 2019-11-29:
Yes, working perfectly after the update.  

Thanks much!!