FOSSASystems/FOSSASAT-1 #7: Not possible to register

mfalkvidd, 2019-11-30:
Sorry if this is not the right place to report a problem for http://groundstationdatabase.com/registerStation.php but I couldn't find any other place.

I am unable to register. The "Optional" Name if the institution seems to be mandatory, and I am not able to enter a valid gps location. The validation message doesn't offer any information on what it considers valid.

[image]

Bambofy, 2019-11-30:
Hello, Thanks for reporting this bug,

I have removed the required flag from the institution input box and made sure it is an optional parameter for the database.

The problem with the GPS coordinates is that we have fixed the precision of the numbers to
0.0001'ths so it must be at most, 4 decimal places long.

Thanks, Richard

mfalkvidd, 2019-11-30:
Great. Would it be possible to let the error message mention that, instead of just saying ”invalid”?

juliantech5, 2020-05-02:
@Bambofy  On my end another suggestion is adding a dashboard for the complied data and separating data in sections (ie: battery, messages, temps etc). Oh and adding FOSSASAT-1B/2 options

Bambofy, 2020-05-02:
> @Bambofy On my end another suggestion is adding a dashboard for the complied data and separating data in sections (ie: battery, messages, temps etc). Oh and adding FOSSASAT-1B/2 options

I will look into this asap