FOSSASystems/FOSSASAT-1 #17: RFM9X series not supported ?

fraab, 2020-10-20:
I've been struggling to get an Adafruit RFM96 module to work with your code.   One issue appears to be with the sync word 0X0F0F.  Since RadioLib's RFM9x support is based on the SX127x, when you dropped support for the SX127x (issue 10), that also dropped support for the RFM9x family.  May I suggest you update the documentation on Github and in your Communications Guide to reflect this fact.

jgromes, 2020-10-20:
I'm not entirely sure what you're trying to do. If you're trying to get the flight code working with SX127x (RFM9x is nearly identical device/silicon), then you're going to run into a lot more trouble than just the sync word, the whole thing is built around SX126x.

If you're trying to get a ground station running with SX127x, I would advise against using the GroundStation sketch in this repository, as FOSSASAT-1 is no longer operational (and likely already de-orbited). There's an updated version of the ground station sketch for [FOSSASAT-1B](https://github.com/FOSSASystems/FOSSASAT-1B/blob/master/software/manual_test/GroundStation/GroundStation.ino) in which the sync word issue has been resolved. However, those sketches were mainly used in development - for actual ground station operation, I highly recommend this project: https://github.com/G4lile0/ESP32-OLED-Fossa-GroundStation

fraab, 2020-10-21:
Jan,

I am attempting to build a ground station. After verifying my hardware
configuration worked with RadioLib's example, I started with your PingPong
program running on 2 seperate systems.
I needed to change SX1262 to RFM96.  Otherwise radio,begin() returned -2
(Chip not found).  I also changed radio.setDio1Action(setFlag) to
radio.setDia0Action(setFlag) in both places.  Doing so, my systems would
transmit but not receive.  (I verified using a UHF radio tuned to 436.7)

When I removed the sync word by changing radio.begin(436.7, 125.0, 11, 8,
0x0F0F) to radio.begin(436.7, 125.0, 11, 8)   the receiver worked as
expected.

Note when specifying the sync word in the begin function, the
compiler returned a warning "large integer implicitly truncated to unsigned
type".

Thanks for pointing me to the ground station code in FOSSASAT-1B as well as
G4lile0's code.  I'll investigate both.
I did a quick compile of your ground station code in FOSSASAT-1B.
FSK_DATA_SHAPPING and FSK_CURRENT_LIMIT  were undefined used in setGFSK().
I haven't tracked that problem yet.  I'm using RadioLib version 4.0.4

Thanks for your help.  I look forward to receiving data from FOSSASAT-1B
once deployed.
Please confirm reception of this email,

Fredric Raab
KK6NOW

jgromes, 2020-10-21:
> Note when specifying the sync word in the begin function, the
compiler returned a warning "large integer implicitly truncated to unsigned
type".

LoRa sync word handling on SX126x changed in RadioLib 3.0.0, to make it slightly harder for people (myself included) to set weird sync word that would be incompatible with SX127x. `SX126x::begin` method now only accepts 1-byte sync word, which is then split into nibbles and corrected to ensure SX127x compatibility. For example, if you set SX126x sync word to 0x12, it gets turned into 0x1424. which is somehow the same sync word as SX127x's 0x12.

If you want to specify the other nibbles, you have to use `SX126x::setSyncWord(sync, control)`, so to set FOSSASAT-1 sync word you'd call `SX126x::setSyncWord(0x00, 0xFF);`. I only added that for FOSSASAT-1 compatibility, and would advise against using that. FOSSASAT-1B and 2 will be using 0x12 sync word and are tested to be compatible with all sub-GHz LoRa transceivers.

> FSK_DATA_SHAPPING and FSK_CURRENT_LIMIT  were undefined used in setGFSK()
Oops, forgot to rename those macros - should be fixed now, thanks.